adafruit/Adafruit_AHTX0 #3: Resolution or Accuracy

rob040, 2020-12-12:
There is a mixup of using sensor **Resolution vs. Accuracy**.

The AHT10 sensor example reports the following:

```
Adafruit AHT10/AHT20 test!
AHT10/AHT20 Found!
------------------------------------
Sensor:       AHTx0_T
Type:         Ambient Temp (C)
Driver Ver:   1
Unique ID:    4129
Min Value:    -40.00
Max Value:    85.00
Resolution:   0.30
------------------------------------

------------------------------------
Sensor:       AHTx0_H
Type:         Relative Humidity (%)
Driver Ver:   1
Unique ID:    4128
Min Value:    0.00
Max Value:    100.00
Resolution:   2.00
------------------------------------
```

The AHT10 temperature **resolution** is 0.01C but its typical **accuracy** is +/- 0.30 C.
The AHT10 RH **resolution** is 0.024% but its typical **accuracy** is +/- 2%.

What is expected of sensor to enter as "resolution" parameter?
I gather that is not made clear, and causes this mixup.

ladyada, 2020-12-12:
yah so we use the android standard 'sensor' definition which only has resolution 
https://developer.android.com/guide/topics/sensors/sensors_overview
not accuracy. i agree it would be best to have both since in many cases they are wildly different. do you want to maybe add 'accuracy' to our adafruit sensor definition at the end of the struct?

rob040, 2020-12-12:
Adding an accuracy parameter would be a possible solution. It would help the unwary to distinguish these. Just too often people mistake resolution for accuracy. However, there is generally no single accuracy number; it comes with a set of conditions on how it is determined. Then there is absolute and relative accuracy (after calibration procedure), repeatability, longterm drift, supply dependency, etc. Manufacturers get away with that by stating "typical" accuracy, some less scrupulous than others.
Resolution looks more clear cut, but actually isn't. Yes, when it reports in 0.01C units, that looks to be the resolution. However, these kind of digital sensors have internal processors converting analog signals, digitizing and linearizing them. The actual step size may be larger than the reporting unit.

This not being a simple answer, __it finally boils down to what you want to achieve__. I think your main goal is introducing programming and electronics to people without that background. Therefore I think it is good to show those that there is a difference between resolution and accuracy and take best available information about these.

ladyada, 2020-12-12:
yeah agreed - i think adding an `accuracy` float is as good as its going to get, given we have to have this code all fit on everything from 8 bit micro's to cortex m7s...by default it will just be set to 0 (unknown) and we can set it for the sensors that can benefit. i think the idea here is just to give some rough idea and also say there's two temperature sensors available, you could pick which one is higher 'quality'?