adafruit/Adafruit_AHTX0 #5: Problem with Adafruit_AHTX0 #53

vanderareosa, 2021-07-14:
trying to use Adafruit_AHTX0

Hardware:
Mac OS 10.13
Arduino IDE 1.8
Board ESP32-C3-DevKitM-1

I'm trying to upload an example to use the AHT10 temperature sensor but I'm getting the following error:

/Users/firemedia1/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/cores/esp32/Arduino.h:107:66: note: suggested alternative: 'GPIO_IN_REG'
#define portInputRegister(port) ((volatile uint32_t*)((port)?GPIO_IN1_REG:GPIO_IN_REG))
^~~~~~~~~~~~
/Users/firemedia1/Dropbox/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.cpp:58:33: note: in expansion of macro 'portInputRegister'
misoPort = (BusIO_PortReg *)portInputRegister(digitalPinToPort(misopin));
^~~~~~~~~~~~~~~~~

ladyada, 2021-07-14:
@me-no-dev heya do you know what changed in 2.0.0?

me-no-dev, 2021-07-14:
Yes, this is C3 particular issue. Since C3 does not have more than 32 GPIOs, it has only one IN/OUT etc. registers. Will fix very soon (before the upcoming 2.0.0 release.

ladyada, 2021-07-14:
all good, thank ya!