adafruit/Adafruit_AHTX0 #9: Code hangs when calling the .begin() initialization method

lukadd16, 2022-03-23:
## Environment
- **Arduino Board:** Arduino Nano Every
- **Arduino IDE Version:** 1.8.16
- **AHTX0 Library Version:** 2.0.1
- **Sensor:** [DHT20](https://www.adafruit.com/product/5183)

## Context

I have a project that requires a temperature and humidity sensor, and after evaluating some of the alternatives I chose the [DHT20 sensor](https://www.adafruit.com/product/5183) (which uses the AHT20 under the hood) that Adafruit sells on their store. Due to space constraints the logical option was to go with the Nano Every microcontroller, one that I am aware uses a different processor than the Arduino UNO but I considered this to be a non-issue due to the software compatibility layer that Arduino includes which emulates the registers of an ATMEGA 328.

## Summary of Issue

Loading up the example sketch (the one titled [`adafruit_aht_test`](https://github.com/adafruit/Adafruit_AHTX0/blob/2d961490f21be7aa01bccd9a460e153f4fcb3207/examples/adafruit_aht_test/adafruit_aht_test.ino)) included with this library, it gets compiled and flashed to the Nano Every's memory without issue (see logs below). Upon opening the Serial Monitor (at 115200 baud to match the baud specified by the sketch), all I see is this message...
```
17:06:51.507 -> Adafruit AHT10/AHT20 demo!
```

...no readings from either sensor, no "AHT10 or AHT20 found" message and no "Could not find AHT? Check wiring" message either. This tells me that none of the code after the call to `aht.begin()` on [line 9](https://github.com/adafruit/Adafruit_AHTX0/blob/2d961490f21be7aa01bccd9a460e153f4fcb3207/examples/adafruit_aht_test/adafruit_aht_test.ino#L9) is being reached, which leads me to believe that the core issue is not with the sketch code itself nor with my wiring (else I would be getting the "check wiring" message after initialization failed), but rather with this library. As a result, I am unable to resolve this issue myself as it is outside the scope of my knowledge and responsibility.

### Arduino IDE Logs
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lukad\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lukad\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lukad\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=on -vid-pid=2341_0058 -ide-version=10816 -build-path C:\Users\lukad\AppData\Local\Temp\arduino_build_769867 -warnings=none -build-cache C:\Users\lukad\AppData\Local\Temp\arduino_cache_808864 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\lukad\Documents\Arduino\libraries\Adafruit_AHTX0\examples\adafruit_aht_test\adafruit_aht_test.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lukad\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lukad\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lukad\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=on -vid-pid=2341_0058 -ide-version=10816 -build-path C:\Users\lukad\AppData\Local\Temp\arduino_build_769867 -warnings=none -build-cache C:\Users\lukad\AppData\Local\Temp\arduino_cache_808864 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\lukad\Documents\Arduino\libraries\Adafruit_AHTX0\examples\adafruit_aht_test\adafruit_aht_test.ino
Using board 'nona4809' from platform in folder: C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7
Using core 'arduino' from platform in folder: C:\Users\lukad\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7
Detecting libraries used...
"C:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino/api/deprecated" "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino" "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\variants\\nona4809" "C:\\Users\\lukad\\AppData\\Local\\Temp\\arduino_build_769867\\sketch\\adafruit_aht_test.ino.cpp" -o nul
Alternatives for Adafruit_AHTX0.h: [Adafruit_AHTX0@2.0.1]
ResolveLibrary(Adafruit_AHTX0.h)
  -> candidates: [Adafruit_AHTX0@2.0.1]
"C:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino/api/deprecated" "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino" "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\variants\\nona4809" "-IC:\\Users\\lukad\\Documents\\Arduino\\libraries\\Adafruit_AHTX0" "C:\\Users\\lukad\\AppData\\Local\\Temp\\arduino_build_769867\\sketch\\adafruit_aht_test.ino.cpp" -o nul
Alternatives for Adafruit_BusIO_Register.h: [Adafruit_BusIO@1.11.3]
ResolveLibrary(Adafruit_BusIO_Register.h)
  -> candidates: [Adafruit_BusIO@1.11.3]
"C:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10816 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino/api/deprecated" "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\cores\\arduino" "-IC:\\Users\\lukad\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.7\\variants\\nona4809" "-IC:\\Users\\lukad\\Documents\\Arduin [... cut ...]
5) Select the Arduino Nano Every as our board

6) Enable registers emulation 

6) Upload the example sketch to the Nano Every

7) Open the Serial Monitor at 115200 baud 

### Sketch Code

https://github.com/adafruit/Adafruit_AHTX0/blob/2d961490f21be7aa01bccd9a460e153f4fcb3207/examples/adafruit_aht_test/adafruit_aht_test.ino#L1-L23

## Things I Have Tried

- Installing an older version of the library
- Using a lower baud
- Triple checked wiring
- Tested a second DHT20 with my Arduino UNO R3 (same result)

ladyada, 2022-03-23:
hiya, questions about how to use adafruit products are best posted in the adafruit forums  at [forums.adafruit.com](forums.adafruit.com/)
thats where tech support is handled - github issues are where folks can issue bug reports (e.g. something that ought to work but doesnt due to a hardware or software bug) since that isnt clearly the issue here and most people are using hte sensor fine, we don't think that's the issue.

thanks! :)

lukadd16, 2022-03-23:
> github issues are where folks can issue bug reports (e.g. something that ought to work but doesnt due to a hardware or software bug)

I wouldn't have opened an issue here if I didn't believe this was the case. I thought the description of my issue was very clear in stating that what I'm doing _should_ work but doesn't due to a factor that is out of my control. Nonetheless, I have started a [thread](https://forums.adafruit.com/viewtopic.php?f=19&t=189626) on the forums as requested so we'll see where this goes.

lukadd16, 2022-04-04:
The forum thread led to me receiving a replacement unit that worked perfectly without issue, hence this GH issue is no longer relevant.

closed 2022-04-04 by lukadd16