adafruit/Adafruit_AHTX0 #11: AHTX0 wont work with ESP-01

keremers, 2023-03-21:
- Arduino board:  ESP-01

- Arduino IDE version (found in Arduino -> About Arduino menu):  Arduino IDE 2.0.4

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
  - Attach AHT10 to ESP-01 (SDA -> GPIO2, SCL -> GPIO0)
  -  Add delay (1000) before first Serial.println. Becasue Serial port is shared with GPIO this time is to wait while serial port being setup.
  - Ad print commands for SDA and SCL definitions of wire.h 
  - Compile https://github.com/adafruit/Adafruit_AHTX0/blob/master/examples/adafruit_aht_test/adafruit_aht_test.ino
  - The output is SDA=4, SCL=5. even if I add new define statements the ATHX0 library assumes these default pins.
  - The program cannot find the sensor and the program hangs.

caternuson, 2023-03-21:
Does it work if you connect to pins 4 and 5?

The I2C pins used for Wire are defined in the ESP-01 Board Support Package, based on whatever you select in Tools->Board menu. They are not defined in this library. If you want to use an alternate Wire bus, the reference can be passed in when calling begin():
https://github.com/adafruit/Adafruit_AHTX0/blob/09b2398cd6380161912303e5b1f90858fa17f78f/Adafruit_AHTX0.h#L83

caternuson, 2023-03-22:
This is a board support issue, not a library issue. It is the board support package that is setting these:

> The output is SDA=4, SCL=5. even if I add new define statements the ATHX0 library assumes these default pins.

closed 2023-08-13 by ladyada